LibreSolar/bms-firmware #2: ToDo: Re-arrange OLED-display output to support max. cell-number

case06, 2017-02-05:
The current version supports up to 12 cells so far.  For showing the output of the max possible number of 15 cells, its necessary to re-arrange the layout, to gain one more row, which is needed to display all 15 cells in 3 columns and 5 rows.   This will be some fizzy work  (i'll do it later, when i deploy my BQ40 board) but should be in general possible on a 128x64 display: we need 7 rows in total, which gives us 9 points per row; that is 8 points font-size/height plus 1 point for a separation line between the rows.

martinjaeger, 2017-02-05:
Maybe not the right place for the 24V version of the board (with 10s bq76930)... either we should rename this repository or open a separate one for the 15s version.

case06, 2017-02-05:
I would strongly recommend/opt for having just one repository for an universal firmware, which supports any possible number of cells just by changing/configuring some settings (otherwise we would generate to much overhead/redundancy/noise).  Even if that means to rename it and i have to do a new fork ;)

But oth i could also live with a "wrong" or misleading name of the software which could be justiifed by "historical growth"  ... and btw., its not to much misleading, because of its subtitle "Firmware for 24/48V BMS board using bq76930 (max. 10s cells) " ... so maybe its ok, to simply change this subtitle towards something like "Firmware for LibreSolar BMS boards using bq76930 (max. 10 cells / 36V) or bq76940 (max 15 cells / 54V)".

case06, 2017-02-05:
I would strongly recommend/opt for having just only one repository for an universal firmware, which supports any possible number of cells just by changing/configuring some settings (otherwise we would generate to much overhead/redundancy/noise). Even if that means to rename it and i have to do a new fork ;)

But oth i could also live with a "wrong" or misleading name of the software which could be justiifed by "historical growth" ... and btw., its not too much misleading, because of its subtitle "Firmware for 24/48V BMS board using bq76930 (max. 10s cells) " ... so maybe its ok, to simply change this subtitle towards something like "Firmware for LibreSolar BMS boards using bq76930 (max. 10 cells / 36V) or bq76940 (max 15 cells / 54V)".

martinjaeger, 2024-01-02:
Fixed with Zephyr driver added in 543827316bd7d9114c6904e40dd5d96239353877.

The OLED uses two separate screens for BMSs with >8 cells.

closed 2024-01-02 by martinjaeger