LibreSolar/bms-firmware #11: ISL94202: Auto-detect number of cells

martinjaeger, 2021-05-12:
The number of cells is currently still hard-coded and set in `bms_init_hardware()` in `bms_isl94202.c`.

The ISL94202 does not automatically detect the number of cells. It has to be written to a register.

Possible algorithm:

- Start with 8 cells enabled.
- Measure voltages and find out actual number of cells (including plausibility-check with pack voltage)
- Set number of cells in chip.

case06, 2021-05-13:
In Dacians SBMS it also has to be set manually by configs.  I wonder if there is any risk for a constellation with e.g. a constellation of unusual low unloaded cells that could perhaps lead to a missinterpretation and wrong celltype-classification ?

martinjaeger, 2021-05-13:
Mhh, true, that could be an issue. But if a cell is below e.g. 1V the pack should not be used anyway anymore...

Today I realized that the default setting of the chip is actually 3 cells, so the algorithm should start from there.