LibreSolar/bms-firmware #15: Make OCV curve configurable via ThingSet

martinjaeger, 2021-05-13:
This allows to better customize the BMS depending on the used cell.

pasrom, 2024-01-25:
@martinjaeger, are you currently working on this issue? If not, I'd like to start on it. It seems like a good addition to our project.

martinjaeger, 2024-01-25:
No, I'm not working on that right now and would appreciate any support.

I have added support for writing arrays via ThingSet in text mode a few days ago, which might be helpful for this. It just needs an update to the latest git SHA for the ThingSet SDK in the west.yml.

pasrom, 2024-01-26:
@martinjaeger I have a follow-up question: Do you mean that each OCV curve should be configurable, or only when `CELL_TYPE_CUSTOM` is selected?

martinjaeger, 2024-01-26:
That's a good question. Probably it would make sense to use the `_CUSTOM` type for that and keep the others as a sensible default if no detailed cell data is available.

We should probably also allow to adjust the corresponding SOC points array, so that the resolution can be increased at very low and high SOCs, where the gradient of the OCV curve is steeper and not so linear.

What do you think?

pasrom, 2024-01-29:
> That's a good question. Probably it would make sense to use the `_CUSTOM` type for that and keep the others as a sensible default if no detailed cell data is available.

Agreed on using `_CUSTOM` type for specific cases and default types for general use.

> We should probably also allow to adjust the corresponding SOC points array, so that the resolution can be increased at very low and high SOCs, where the gradient of the OCV curve is steeper and not so linear.

Also in favor of adjusting the SOC points array for better resolution at extreme SOCs.

PR #49 (2024-02-01): Make OCV & SoC points configurable via ThingSet
This PR proposes a solution to fix #15. This includes using the new feature of the ThingSet text mode to add the OCV & SoC points via ThingSet to the config. Additionally, the SoC points will be moved into `bms_common` to harmonize with the OCV points.
closed 2024-02-01 by martinjaeger