LibreSolar/bms-firmware #17: BQ76952

kidspro88, 2021-07-23:
Hi @martinjaeger ,

Do you have any plan to support BQ76952? 
---------------------------
The BQ76952 has several differences from the BQ76940. The voltage and current measurement accuracy are improved, the measurement speed is much faster, the BQ76952 drives high-side FETs, it supports additional communication protocols, it includes automated cell balancing and automated FET control to recover from protection faults. If the BQ76940 meets your needs, it is a good device and may be simpler to understand its limited features. The BQ76952 is our newest industrial battery monitor with better performance and features. 

TI

martinjaeger, 2021-07-23:
Yes, I am actually planning to develop a new BMS based on exactly that IC, but still need to secure some funding ;)

kidspro88, 2021-07-24:
Yep, sure, About the sponsor, can you contact me via e-mail?

martinjaeger, 2021-07-26:
@kidspro88 I have dropped you an email on Friday (to the address I found in your commits, as there is no public email on your profile).

Let me know if you didn't receive it.

martinjaeger, 2021-07-29:
@kidspro88 the mail I sent you just came back because of full quota. Can you please let me know your email address or contact me via martin at libre.solar?

martinjaeger, 2021-07-30:
We are currently applying for funding from EnAccess foundation to develop a BMS based on the bq76952. For anyone interested in such a BMS, please leave some feedback here: https://community.enaccess.org/t/open-source-flexible-battery-management-system-for-off-grid-energy-storage/75

The more feedback we get, the more likely it will be to make it happen.

case06, 2021-07-30:
For beeing able to comment there you mus share an invite.

There is no other way to registrate there.

martinjaeger, 2021-07-30:
> For beeing able to comment there you must share an invite.

Oops, didn't know that. A bit annoying for a community... I will send an invite to your email.

martinjaeger, 2022-01-10:
The bq76952 is already partly supported now and we are continuing to fully integrate it. Closing this issue.

closed 2022-01-10 by martinjaeger