LibreSolar/bms-firmware #20: Change settings over Thingset, settings are not stored in EEPROM

xsider, 2021-08-29:
Tried to save settings over thingset (serial) in EEPROM failed. After reboot, EEPROM still seems to be empty. Tested on current git main trunk. 

Workflow: started with a new EEPROM, tried to change config parameter over serial thingset connection. After parameter changed, reboot the hardware, readback config

Debug terminal output, no EEPROM mounted (recognized correctly):
```
I: Init of CAN_1 done
*** Booting Zephyr OS build zephyr-v20500  ***
Hardware: Libre Solar BMS 5S50 SC (v0.1)
Firmware: unknown
E: failed to read EEPROM (err -5)
EEPROM: read error: -5
```
Soldered EEPROM, first boot:
```
*** Booting Zephyr OS build zephyr-v20500  ***                                  
Hardware: Libre Solar BMS 5S50 SC (v0.1)                                        
Firmware: unknown                                                               
EEPROM: Empty or data layout version changed
```

Read config and change parameter over serial Thingset connection, then reset device:
```
Received Request (5 bytes): ?conf                                               
Received Request (24 bytes): =conf {"BatNom_Ah":60.0}                           
EEPROM: Data successfully stored.                                               
Received Request (11 bytes): !exec/reset 
```

After reboot, EEPROM still seems to be not initialized, readback of conf returns default parameter, not changed one.
```
...
EEPROM: Empty or data layout version changed 
Received Request (5 bytes): ?conf
```
The explicit command for save settings does not work either.
```
Received Request (24 bytes): =conf {"BatNom_Ah":80.0}                           
EEPROM: Data successfully stored.                                               
Received Request (5 bytes): ?conf                                               
Received Request (19 bytes): !exec/save-settings                                
EEPROM: Data successfully stored.
Received Request (11 bytes): !exec/reset 
I: Init of CAN_1 done                                                           
*** Booting Zephyr OS build zephyr-v20500  ***                                  
Hardware: Libre Solar BMS 5S50 SC (v0.1)                                        
Firmware: unknown                                                               
EEPROM: Empty or data layout version changed 
```

martinjaeger, 2021-08-29:
Thanks for reporting this bug. During my latest tests with BMS 8s50 it worked fine as far as I remember. So I'll check again wit BMS 5s50 next week.

martinjaeger, 2021-09-02:
Didn't have the time to double-check this issue yet, but can you tell me which EEPROM chip you are using exactly? Is it the 24AA01? I'm asking because if you took a larger one (e.g. the 24AA32) the address width and page size have to be adjusted in the DTS. See also comments in the [DTS of MPPT 2420 LC](https://github.com/LibreSolar/charge-controller-firmware/blob/main/boards/arm/mppt_2420_lc/mppt_2420_lc.dts).

xsider, 2021-09-02:
i use a 24AA32, i had dismissed 24AA01 to reduce stockpile and the price difference is small. i will try the DTS settings for the correct page size.

martinjaeger, 2021-10-04:
@xsider did you double-check with correct page size and can we consider the issue resolved?

xsider, 2021-11-01:
> @xsider did you double-check with correct page size and can we consider the issue resolved?

This issue can be closed. It worked with your recommended settings. Data is stored in EEPROM.

I compiled with DTS-settings for 24AA32 bms_5s50_sc.dts:
```
eeprom@50 {
    compatible = "atmel,at24";
    reg = <0x50>;
    label = "EEPROM_0";
    size = <32768>;
    pagesize = <32>;
    address-width = <16>; 
    timeout = <7>;
};

```

closed 2021-11-01 by martinjaeger