LibreSolar/bms-firmware #27: project build issue

1278043638, 2022-09-20:
I want to test this code, but I can't create it. I successfully tested other similar codes, but this code is not. Is it caused by the environment? Here is my error code. What can you do to help me solve it? Thank you!!!
FAILED: zephyr/CMakeFiles/zephyr.dir/lib/os/onoff.c.obj
FAILED: zephyr/CMakeFiles/zephyr.dir/soc/riscv/esp32c3/soc_irq.S.obj
FAILED: zephyr/CMakeFiles/zephyr.dir/lib/os/mpsc_pbuf.c.obj
FAILED: zephyr/CMakeFiles/zephyr.dir/soc/riscv/esp32c3/soc.c.obj
FATAL ERROR: command exited with status 1: 'C:\Program Files\CMake\bin\cmake.EXE' --build 'C:\Users\12780\zephyrproject\bms-firmware\app\build'

martinjaeger, 2022-09-20:
Sorry, but I need more details about what you have been trying to do in order to help you.

Please provide some information about the commands used to build the firmware, the documentation you read and where exactly you see this error.

1278043638, 2022-09-20:
I am developing a 48V electric vehicle BMS, which has been developed on the STM32 before. The ESP32 is my first time to use it.I just installed the zephyr toolchain and tried to build the firmware.
The build instruction I use is "west build -b bms_16s100_sc_esp32c3".However, some register instructions are unrecognized.
The following instructions.
C:/Users/12780/zephyrproject/zephyr/include/zephyr/arch/riscv/arch.h:244: Error: unrecognized opcode `csrrc s0,mstatus,8'
C:/Users/12780/zephyrproject/zephyr/include/zephyr/arch/riscv/arch.h:258: Error: unrecognized opcode `csrs mstatus,s1'
C:/Users/12780/zephyrproject/zephyr/soc/riscv/esp32c3/soc.c:47: Error: unrecognized opcode `csrw mtvec,t0'
C:/Users/12780/zephyrproject/zephyr/soc/riscv/esp32c3/soc.c:52: Error: unrecognized opcode `csrrc a5,mstatus,8'
C:/Users/12780/zephyrproject/zephyr/soc/riscv/esp32c3/soc.c:108: Error: unrecognized opcode `csrrc a5,mstatus,8'
This is the last error
FATAL ERROR: command exited with status 1: 'C:\Program Files\CMake\bin\cmake.EXE' --build 'C:\Users\12780\zephyrproject\bms-firmware\app\build'

martinjaeger, 2022-09-20:
I've never seen such an error message before. Are you able to build the [Zephyr hello world sample for the ESP32-C3](https://docs.zephyrproject.org/3.1.0/boards/riscv/esp32c3_devkitm/doc/index.html)?

1278043638, 2022-10-09:
Yes, I can build the Zephyr hello world sample for the ESP32-C3.
I (49) boot: ESP-IDF 2cb20ac6c 2nd stage bootloader
I (49) boot: compile time 15:17:35
I (49) boot: chip revision: 3
I (51) boot.esp32c3: SPI Speed      : 40MHz
I (56) boot.esp32c3: SPI Mode       : DIO
I (61) boot.esp32c3: SPI Flash Size : 4MB
I (65) boot: Enabling RNG early entropy source...
I (71) boot: Partition Table:
I (74) boot: ## Label            Usage          Type ST Offset   Length
I (82) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (89) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (96) boot:  2 factory          factory app      00 00 00010000 00100000
I (104) boot: End of partition table
I (108) esp_image: segment 0: paddr=00010020 vaddr=00000020 size=0001ch (    28)
I (116) esp_image: segment 1: paddr=00010044 vaddr=3fc83c30 size=00078h (   120) load
I (125) esp_image: segment 2: paddr=000100c4 vaddr=3fc83ca8 size=00200h (   512) load
I (133) esp_image: segment 3: paddr=000102cc vaddr=40380000 size=0288ch ( 10380) load
I (144) esp_image: segment 4: paddr=00012b60 vaddr=00000000 size=0d4d8h ( 54488)
I (162) esp_image: segment 5: paddr=00020040 vaddr=3c000040 size=00588h (  1416) map
I (163) esp_image: segment 6: paddr=000205d0 vaddr=00000000 size=0fa48h ( 64072)
I (182) esp_image: segment 7: paddr=00030020 vaddr=42010020 size=031f0h ( 12784) map
I (186) boot: Loaded app from partition at offset 0x10000
I (186) boot: Disabling RNG early entropy source...
�*** Booting Zephyr OS build v3.2.0-rc2-92-g6030b85f71ce  ***
Hello World! esp32c3_devkitm

martinjaeger, 2022-10-09:
I have updated the firmware for Zephyr v3.2, which contained lots of improvements for ESP32-C3.

Can you test to build that? It's in the `develop` branch.

martinjaeger, 2022-10-10:
That sounds great. When you build the BMS make sure that you use the new BMS C1 design (which is based on the previous bms_16s100_sc_esp32c3 schematic, but uses a single PCB only).

Also I'm currently facing an issue with I2C comms on ESP32-C3. It's most probably a software issue, but you may want to wait until it's fixed before you order something.

Really appreciate that you are planning to contribute back your findings to improve the project!

closed 2022-10-24 by 1278043638