LibreSolar/bms-firmware #29: The board stops working

1278043638, 2022-11-07:
I finished a board last week and is testing its performance. But it's strange. It starts for a while and then reports an error. After it restarts, it will be fine again. I can't understand this error. Can you help me to read it? In addition, I found that when it was just started, there would be no detection data. The following is my error message:
#mSerial {"rBat_V":8.46,"rBat_A":0.00,"rBat_degC":-58.8,"rSOC_pct":0.0,"rErrorFlags":2,"rBmsState":2,"rCells_V":[0.651,0.136,0.124,0.350,1.058,6.060,0.000,0.000,0.000,0.000,0.000,0.000,0.000,0.000,0.000,0.000],"rCellAvESP-ROM:esp32c3-api1-20210207
Build:Feb  7 2021
rst:0x15 (USB_UART_CHIP_RESET),boot:0xf (SPI_FAST_FLASH_BOOT)
Saved PC:0x4201ac50
SPIWP:0xee
mode:DIO, clock div:2
load:0x3fcd6100,len:0x15e4
load:0x403ce000,len:0x920
load:0x403d0000,len:0x2c4c
SHA-256 comparison failed:
Calculated: f92edd0762896a9fa673795fc470e6fa77cf833e5c0950e4978b697f167c1bb4
Expected: 5575fe58924c1e08c5eadfa2ba7862131ebe09b778e446c7d1ed87641a26bbfa
Attempting to boot anyway...
entry 0x403ce000
[0;32mI (43) boot: ESP-IDF 2cb20ac6c 2nd stage bootloader[0m
[0;32mI (43) boot: compile time 14:42:43[0m
[0;32mI (43) boot: chip revision: 3[0m
[0;32mI (45) boot.esp32c3: SPI Speed      : 40MHz[0m
[0;32mI (49) boot.esp32c3: SPI Mode       : DIO[0m
[0;32mI (54) boot.esp32c3: SPI Flash Size : 4MB[0m
[0;32mI (59) boot: Enabling RNG early entropy source...[0m
[0;32mI (64) boot: Partition Table:[0m
[0;32mI (68) boot: ## Label            Usage          Type ST Offset   Length[0m
[0;32mI (75) boot:  0 nvs              WiFi data        01 02 00009000 00006000[0m
[0;32mI (83) boot:  1 phy_init         RF data          01 01 0000f000 00001000[0m
[0;32mI (90) boot:  2 factory          factory app      00 00 00010000 00100000[0m
[0;32mI (98) boot: End of partition table[0m
[0;32mI (102) esp_image: segment 0: paddr=00010020 vaddr=00000020 size=0001ch (    28) [0m
[0;32mI (110) esp_image: segment 1: paddr=00010044 vaddr=3fc88d80 size=00758h (  1880) load[0m
[0;32mI (119) esp_image: segment 2: paddr=000107a4 vaddr=3fc894d8 size=00418h (  1048) load[0m
[0;32mI (127) esp_image: segment 3: paddr=00010bc4 vaddr=40380000 size=05160h ( 20832) load[0m
[0;32mI (141) esp_image: segment 4: paddr=00015d2c vaddr=00000000 size=0a30ch ( 41740) [0m
[0;32mI (153) esp_image: segment 5: paddr=00020040 vaddr=3c000040 size=02724h ( 10020) map[0m
[0;32mI (155) esp_image: segment 6: paddr=0002276c vaddr=00000000 size=0d8ach ( 55468) [0m
[0;32mI (172) esp_image: segment 7: paddr=00030020 vaddr=42010020 size=16cf8h ( 93432) map[0m
[0;32mI (195) boot: Loaded app from partition at offset 0x10000[0m
[0;32mI (195) boot: Disabling RNG early entropy source...[0m
*** Booting Zephyr OS build zephyr-v3.2.0-9-g8625a40ec488  ***
Hardware: Libre Solar BMS C1 (v0.3)

Firmware: 

E: BMS hardware initialization failed, retrying in 10s
#mSerial {"rBat_V":0.00,"rBat_A":0.00,"rBat_degC":0.0,"rSOC_pct":0.0,"rErrorFlags":0,"rBmsState":0,"rCells_V":[0.000,0.000,0.000,0.000,0.000,0.000,0.000,0.000,0.000,0.000,0.000,0.000,0.000,0.000,0.000,0.000],"rCellAvg_V":0.000,"rCellMin_V":0.000,"rCellMax_V":0.000,"rBalancingStatus":0}#mSerial {"rBat_V":0.00,"rBat_A":0.00,"rBat_degC":0.0,"rSOC_pct":0.0,"rErrorFlags":0,"rBmsState":0,"rCells_V":[0.000,0.000,0.000,0.000,0.000,0.000,0.000,0.000,0.000,0.000,0.000,0.000,0.000,0.000,0.000,0.000],"rCellAvg_V":0.000,"rCellMin_V":0.000,"rCellMax_V":0.000,"rBalancingStatus":0}#mSerial {"rBat_V":0.00,"rBat_A":0.00,"rBat_degC":0.0,"rSOC_pct":0.0,"rErrorFlags":0,"rBmsState":0,"rCells_V":[0.000,0.000,0.000,0.000,0.000,0.000,0.000,0.000,0.000,0.000,0.000,0.000,0.000,0.000,0.000,0.000],"rCellAvg_V":0.000,"rCellMin_V":0.000,"rCellMax_V":0.000,"rBalancingStatus":0}#mSerial {"rBat_V":0.00,"rBat_A":0.00,"rBat_degC":0.0,"rSOC_pct":0.0,"rErrorFlags":0,"rBmsState":0,"rCells_V":[0.000,0.000,0.000,0.000,0.000,0.000,0.000,0.000,0.000,0.000,0.000,0.000,0.000,0.000,0.000,0.000],"rCellAvg_V":0.000,"rCellMin_V":0.000,"rCellMax_V":0.000,"rBalancingStatus":0}#mSerial {"rBat_V":0.00,"rBat_A":0.00,"rBat_degC":0.0,"rSOC_pct":0.0,"rErrorFlags":0,"rBmsState":0,"rCells_V":[0.000,0.000,0.000,0.000,0.000,0.000,0.000,0.000,0.000,0.000,0.000,0.000,0.000,0.000,0.000,0.000],"rCellAvg_V":0.000,"rCellMin_V":0.000,"rCellMax_V":0.000,"rBalancingStatus":0}#mSerial {"rBat_V":0.00,"rBat_A":0.00,"rBat_degC":0.0,"rSOC_pct":0.0,"rErrorFlags":0,"rBmsState":0,"rCells_V":[0.000,0.000,0.000,0.000,0.000,0.000,0.000,0.000,0.000,0.000,0.000,0.000,0.000,0.000,0.000,0.000],"rCellAvg_V":0.000,"rCellMin_V":0.000,"rCellMax_V":0.000,"rBalancingStatus":0}#mSerial {"rBat_V":0.00,"rBat_A":0.00,"rBat_degC":0.0,"rSOC_pct":0.0,"rErrorFlags":0,"rBmsState":0,"rCells_V":[0.000,0.000,0.000,0.000,0.000,0.000,0.000,0.000,0.000,0.000,0.000,0.000,0.000,0.000,0.000,0.000],"rCellAvg_V":0.000,"rCellMin_V":0.000,"rCellMax_V":0.000,"rBalancingStatus":0}#mSerial {"rBat_V":0.00,"rBat_A":0.00,"rBat_degC":0.0,"rSOC_pct":0.0,"rErrorFlags":0,"rBmsState":0,"rCells_V":[0.000,0.000,0.000,0.000,0.000,0.000,0.000,0.000,0.000,0.000,0.000,0.000,0.000,0.000,0.000,0.000],"rCellAvg_V":0.000,"rCellMin_V":0.000,"rCellMax_V":0.000,"rBalancingStatus":0}#mSerial {"rBat_V":0.00,"rBat_A":0.00,"rBat_degC":0.0,"rSOC_pct":0.0,"rErrorFlags":0,"rBmsState":0,"rCells_V":[0.000,0.000,0.000,0.000,0.000,0.000,0.000,0.000,0.000,0.000,0.000,0.000,0.000,0.000,0.000,0.000],"rCellAvg_V":0.000,"rCellMin_V":0.000,"rCellMax_V":0.000,"rBalancingStatus":0}#mSerial {"rBat_V":0.00,"rBat_A":0.00,"rBat_degC":0.0,"rSOC_pct":0.0,"rErrorFlags":0,"rBmsState":0,"rCells_V":[0.000,0.000,0.000,0.000,0.000,0.000,0.000,0.000,0.000,0.000,0.000,0.000,0.000,0.000,0.000,0.000],"rCellAvg_V":0.000,"rCellMin_V":0.000,"rCellMax_V":0.000,"rBalancingStatus":0}I: detected bq device number: 0x7695
#mSerial {"rBat_V":8.45,"rBat_A":0.05,"rBat_degC":-58.8,"rSOC_pct":0.0,"rErrorFlags":322,"rBmsState":0,"rCells_V":[0.661,0.144,0.129,0.361,1.067,6.060,-0.001,0.000,-0.001,0.000,0.000,0.000,0.000,0.000,0.000,0.000],"rCellAvg_V":2.596,"rCellMin_V":0.661,"rCellMax_V":6.060,"rBalancingStatus":0}#mSerial {"rBat_V":8.44,"rBat_A":0.03,"rBat_degC":-58.8,"rSOC_pct":0.0,"rErrorFlags":322,"rBmsState":0,"rCells_V":[0.657,0.136,0.125,0.358,1.067,6.060,-0.001,0.000,0.000,0.000,0.000,0.000,0.000,0.000,0.000,0.000],"rCellAvg_V":2.595,"rCellMin_V":0.657,"rCellMax_V":6.060,"rBalancingStatus":0}#mSerial {"rBat_V":8.45,"rBat_A":0.03,"rBat_degC":-58.8,"rSOC_pct":0.0,"rErrorFlags":322,"rBmsState":0,"rCells_V":[0.545,0.221,0.141,0.318,0.944,6.060,0.000,0.000,-0.001,0.000,0.000,0.000,0.000,0.000,0.000,0.000],"rCellAvg_V":2.516,"rCellMin_V":0.545,"rCellMax_V":6.060,"rBalancingStatus":0}#mSerial {"rBat_V":8.45,"rBat_A":0.05,"rBat_degC":-58.8,"rSOC_pct":0.0,"rErrorFlags":322,"rBmsState":0,"rCells_V":[0.630,0.173,0.130,0.355,1.046,6.060,-0.001,0.000,0.000,0.000,0.000,0.000,0.000,0.000,0.000,0.000],"rCellAvg_V":2.579,"rCellMin_V":0.630,"rCellMax_V":6.060,"rBalancingStatus":0}#mSerial {"rBat_V":8.46,"rBat_A":0.03,"rBat_degC":-58.8,"rSOC_pct":0.0,"rErrorFlags":322,"rBmsState":0,"rCells_V":[0.650,0.146,0.118,0.347,1.046,6.060,-0.001,0.001,-0.001,0.000,0.000,0.000,0.000,0.000,0.000,0.000],"rCellAvg_V":2.585,"rCellMin_V":0.650,"rCellMax_V":6.060,"rBalancingStatus":0}#mSerial {"rBat_V":8.45,"rBat_A":-0.03,"rBat_degC":-58.8,"rSOC_pct":0.0,"rErrorFlags":322,"rBmsState":0,"rCells_V":[0.654,0.137,0.122,0.354,1.061,6.060,0.000,0.000,-0.001,0.000,0.000,0.000,0.000,0.000,0.000,0.000],"rCellAvg_V":2.592,"rCellMin_V":0.654,"rCellMax_V":6.060,"rBalancingStatus":0}#mSerial {"rBat_V":8.45,"rBat_A":0.03,"rBat_degC":-58.8,"rSOC_pct":0.0,"rErrorFlags":322,"rBmsState":0,"rCells_V":[0.665,0.143,0.128,0.356,1.057,6.060,-0.001,0.001,0.000,0.000,0.000,0.000,0.000,0.000,0.000,0.000],"rCellAvg_V":2.594,"rCellMin_V":0.665,"rCellMax_V":6.060,"rBalancingStatus":0}#mSerial {"rBat_V":8.45,"rBat_A":0.00,"rBat_degC":-58.8,"rSOC_pct":0.0,"rErrorFlags":322,"rBmsState":0,"rCells_V":[0.552,0.216,0.140,0.322,0.950,6.060,-0.001,0.000,0.000,0.000,0.000,0.000,0.000,0.000,0.000,0.000],"rCellAvg_V":2.521,"rCellMin_V":0.552,"rCellMax_V":6.060,"rBalancingStatus":0}#mSerial {"rBat_V":8.45,"rBat_A":0.00,"rBat_degC":-58.8,"rSOC_pct":0.0,"rErrorFlags":322,"rBmsState":0,"rCells_V":[0.627,0.170,0.129,0.353,1.039,6.060,0.000,0.000,-0.001,0.000,0.000,0.000,0.000,0.000,0.000,0.000],"rCellAvg_V":2.575,"rCellMin_V":0.627,"rCellMax_V":6.060,"rBalancingStatus":0}#mSerial {"rBat_V":8.45,"rBat_A":0.03,"rBat_degC":-58.8,"rSOC_pct":0.0,"rErrorFlags":3 [... cut ...]
E:  mcause: 1, Instruction Access fault
E:   mtval: e69b8b86
E:      a0: d196c8c4    t0: cf478fcb
E:      a1: db5067ed    t1: 616bb785
E:      a2: a1c1f5a4    t2: ae13db79
E:      a3: 1deda449    t3: 128a90d2
E:      a4: b551c932    t4: f0dbeb04
E:      a5: e5070e22    t5: 54ec9f26
E:      a6: c89959b8    t6: 4b3fe0f8
E:      a7: 50068842
E:      ra: 3fc88400
E:    mepc: e69b8b86
E: mstatus: 00000091
E: 
E: >>> ZEPHYR FATAL ERROR 0: CPU exception on CPU 0
E: Current thread: 0x3fc85760 (main)
E: Halting system

martinjaeger, 2022-11-07:
Nice that your rebuilt the board and got it set up so far.

The issue with no data being read immediately after start-up is because of the glitches described here: https://github.com/zephyrproject-rtos/zephyr/issues/51250
I haven't found a solution other than trying again after a few seconds, yet.

The Instruction Access Fault looks strange. I've never seen that issue. Which exact firmware commit are you using? And did you change anything in the firmware so far? I'll try to reproduce the error.

1278043638, 2022-11-07:
I tried again. It happened by accident with a high frequency. I found this problem when I pressed S1 button for 3 seconds, then stopped, and then pressed S1 to start. Occasionally, it will get stuck and no error will occur. If it reports an error, such an error will occur.

1278043638, 2022-11-07:
I seem to understand that it is the error of the computer serial port receiving data. When I plug and unplug the USB cable again, it returns to normal.

martinjaeger, 2023-01-09:
@1278043638 With the release of today all issues with the ESP32-C3 serial port should be resolved. Make sure to run `west update` after pulling the BMS firmware updates.

closed 2023-01-13 by 1278043638