LibreSolar/bms-firmware #30: Inaccurate battery voltage detection accuracy

1278043638, 2022-11-09:
During the test, I found that the error of battery 1 was a little large. The voltage measured by the multimeter is 3.83V. How should this be adjusted?

{"rBat_V":22.38,"rBat_A":0.00,"rBat_degC":-58.8,"rSOC_pct":100.0,"rErrorFlags":320,"rBmsState":0,"rCells_V":[3.696,3.826,3.819,3.828,3.829,3.821,0.000,0.000,0.000,0.000,0.000,0.000,0.000,0.000,0.000,0.000],"rCellAvg_V":3.803,"rCellMin_V":3.696,"rCellMax_V":3.829,"rBalancingStatus":0}

martinjaeger, 2022-11-11:
How did you connect the first cell? Did you add two separate wires for GND and C0? If not, you could get a measurement error because the internal power supply current goes through the cell balancing/measurement wire.

I have verified with my hardware and the cell voltages are all very accurate (including the first one). Fluke measurement is same as cell voltage measurement through BMS with deviation of +-1 mV.

1278043638, 2022-11-14:
Thank you for your help. After I checked the schematic diagram, I found that I did not connect bat -, which would cause problems in the measurement. I have solved the problem, and the result is very accurate.

closed 2022-11-14 by 1278043638