LibreSolar/bms-firmware #34: "Libre Solar BMS C1 Firmware Update: LEDs Turn Off, 'No Serial Data Received' Error, and Flashing Problems"

Ananyaaynana, 2023-11-11:
Issue:
I tried updating to the latest firmware version "v23.1-1-g55c2ea2-dirty," but encountered two issues:
The red and green LEDs switch off and get the error "A fatal error occurred: No serial data received." when attempting to flash the code and the LEDs can only be turned on using the S2 button.

Steps Taken:
I created an overlay file (as in https://libre.solar/bms-firmware/src/dev/customization.html) in the app folder with the following code:
{
chosen {
thingset,serial = &usb_serial;
};
}

I  then built and flashed the code using the "west flash" command. When attempting to monitor the board using the "west espressif monitor" command, I encountered the following output:
[image]

Additionally, when I tried to flash the code again, red and green leds turn off but there is voltage at the bat+, bat-, pack+ and pack-. 
[image]

When the leds are off the "west espressif monitor"command gives this:
[image]

On exiting the espressif monitor not able to flash code again. On holding down the S2 button the leds turn back on after blinking for a few seconds.

Any guidance on resolving these issues. Specifically, I need help with:

1) Understanding why the code flashing is not happening and why the LEDs are turning off.

Additional Information:

Board Model:  Libre Solar BMS C1 (v0.3)
Firmware Version: v23.1-1-g55c2ea2-dirty.

martinjaeger, 2023-11-11:
Did you keep the serial monitoring on while you tried to flash (espressif monitor or CuteCom)? That's not possible, as flashing uses the same serial interface.

Ananyaaynana, 2023-11-11:
No the serial monitor was off when flashing the code.

martinjaeger, 2023-11-12:
Ok, that's strange and I can't reproduce it here.

It is expected that the MOSFETs stay on during flashing, as the bq chip operates independent of the MCU.

From what I understood you were finally able to flash new firmware after pressing reset, right? Does it always only work after pressing reset or was this just a single time flashing failed?

Ananyaaynana, 2023-11-13:
No I am still not able to flash code into the board. It seems to stop flashing when trying to change the baud rate.

martinjaeger, 2023-11-13:
Does the Espressif USB JTAG appear in `lsusb`?

If not, maybe something went wrong and you have to manually put the board into bootloader mode to reflash: Connect TP11 to GND and press reset or power cycle the board.

Ananyaaynana, 2023-11-15:
Yes it does appear in `lsusb` as "Espressif USB JTAG/serial debug unit".

Ananyaaynana, 2023-11-15:
[image]
I entered bootloader manually by grounding TP11 and pressed the reset. I was able to flash the code successfully. 
On using the `west espressif monitor` it stops at "waiting for download".
The LED's are still not on after flashing is done.

martinjaeger, 2023-11-15:
Did you disconnect TP11 from GND again?

martinjaeger, 2023-11-15:
I'm almost out of ideas.

Did you run `west update` after updating to the new BMS firmware?

Did you see any compiler warnings?

Can you go back to the old firmware revision and successfully flash / use that?

Is the behavior the same on both boards?

Did you try to delete the firmware directory and get a clean workspace again following the steps in the docs?

Ananyaaynana, 2023-11-16:
I deleted the firmware directory and followed the steps in the documentation and reflashed the board and it worked. Thank you for the help.

closed 2023-11-16 by Ananyaaynana