LibreSolar/bms-firmware #36: LiPo Float Charge & SOC settings

intrinsicaly, 2023-12-20:
Hi Martin,
Thanks for all your work on this project.
Just a couple of question;
1. Does the firmware support a float mode setting?
Float mode with say an 8 hour timer would allow the mppt to stay on and feed any loads without reseting to a bulk charge. This would also pretty much guarantee that the batteries are fully charged, despite any usage throughout the day, when the sun has set.

2. Is the state of charge (SOC) configurable so that 100% can be set at say 3.45V/cell?

Cheers

martinjaeger, 2023-12-20:
Did you mean to ask this in the BMS repo?

The BMS does not really specify any charging mode. It can only make sure that the batteries are not operated outside their allowed operating conditions (voltage, current, temperature) and switch off in case the charger doesn't respect the operating range.

intrinsicaly, 2023-12-20:
Hi Martin,
Please don't think that the following is telling you want to do, just my understanding so far.

I am very new to all of this, but my understanding so far is that the BMS will monitor all of the cells for voltage, current etc (SOC). and mange the overall pack on an individual cell level. Once each cell in the pack is at the same state making the pack 100% charged then the BMS would communicate with the charger/mppt to limit its function or turns it off to protect the batteries. If the BMS gets a float setting this would communicate to limit the charging but not turn it off allowing loads to be supplied by the solar even if the batteries are full. This ensures that the batteries are in the best possible state when the sun sets.

My understanding so far comes from various sources but the video at the following youtube link will hopefully explain better where I coming from - https://www.youtube.com/watch?v=LDXcx5CunP0.

Cheers

DonP2748, 2024-01-08:
Hi Martin and Intrinsicaly,  i think you(@intrinsicaly) might need fuel gauss ics which are not in Martin's BMS project (bq769x0, bq769x2 and ISL94202 look like charger ics and have no estimated SoC, SoH and SoP functionality ?). you might use a fuel gauss ic or use a battery monitor ic (which have both fuel gauss and charger) to read capacity by stm32 and stm32 will send off FET command to charger ic at the specific voltage. 
 
By the way, can I ask you(@martinjaeger ) why not use typedef for enum :D?
Cheers