LibreSolar/bms-firmware #40: Function `bms_soc_update` no longer called post-zephyr bms driver update

pasrom, 2024-01-19:
After the recent Zephyr driver update (commit https://github.com/LibreSolar/bms-firmware/commit/f0ff664c2687e68f9526ed6f4d2f659b1bb0fa19), the `bms_soc_update` function seems to have stopped being invoked. This has led to the State of Charge (SoC) no longer being calculated. Prior to this update, the function was regularly called at this location: https://github.com/LibreSolar/bms-firmware/blob/39bbaeaf668e310dca520255687771dbc1d52ff8/app/src/bq769x2/bms_bq769x2.c#L154-L162

Could you advise on the best location for invoking `bms_soc_update` in the current code framework to maintain precise SoC measurements?

pasrom, 2024-01-19:
I recommend placing the call to `bms_soc_update(&bms)` before line 66, where `bms_state_machine(&bms)` is called.
https://github.com/LibreSolar/bms-firmware/blob/f0ff664c2687e68f9526ed6f4d2f659b1bb0fa19/app/src/main.cpp#L61-L66

martinjaeger, 2024-01-19:
Thanks for raising the issue. I'm planning to do some further BMS testing tomorrow anyway, so I'll have al look into that.

martinjaeger, 2024-01-20:
I agree with your proposed fix and have created a PR (see link above).

commit 67a10e17 (2024-01-20): BMS app: Call SOC update function regularly

Fixes #40
PR #41 (2024-01-20): BMS app: Call SOC update function regularly
Fixes #40

I'm working on some improvements to the test framework, which will also include the SOC calculation. But that's not quite ready, yet. So let's first fix this bug.
commit 77345691 (2024-01-20): BMS app: Call SOC update function regularly

Fixes #40
closed 2024-01-20 by martinjaeger