LibreSolar/bms-firmware #44: QEMU Build Target

seandheath, 2024-01-26:
I'm working on a project where I would like to be able to use a VM to run this BMS. I know ZephyrOS supports QEMU VMs as targets, but I'm not sure of the steps required to get a LibreSolar BMS VM. I tried just building it using `west build -b qemu_x86` but a series of errors from `#define` statements in `devicetree.h` cause the build to fail. I'm going to start working through the errors and see how far I get but wanted to check on here to see if anyone already has some knowledge that might help me out. Thanks!

martinjaeger, 2024-01-26:
Hi @seandheath,

That sound like a cool idea. I haven't tried it so far, but running the BMS in QEMU should be possible without too much work. So far we are only using `native_posix` for testing on a Linux host.

Most probably your Devicetree issues are because you need an overlay for the features necessary for the BMS. Have a look at [tests/boards/native_posix.overlay](https://github.com/LibreSolar/bms-firmware/blob/160089a7b5d2c23e06113a51696707e653d491dc/tests/boards/native_posix.overlay) as a starting point.

The recent refactoring should have made it easier to run with qemu, as we are now following the typical Zephyr way of doing things more closely (especially for the BMS IC drivers).

For the bq769x2 driver we have an emulation layer, which is used for the unit tests. This will also be helpful for qemu. With the emulated driver it should at least be possible to build and run the firmware successfully without actual hardware. The emulation does not simulate the behaviour of the actual chip, though. It just provides in interface to read/write its register contents, which can then be accessed through the `bms_ic` API.

martinjaeger, 2024-04-08:
@seandheath I have just added a `native_sim` build target to allow running the BMS application on a Linux host.

Commit: c921b4d3ca64c0e19b41f9f452df2d382e0c5b7e

Dawson-Hsu, 2024-12-05:
@martinjaeger  I'm try to build the C1 target , but always prompts failure ,The error message is as follows. How should I resolve it?   Or is there a more detailed Workspace setup introduction  Tks !
[image]

Faith-cmyk, 2025-01-06:
Hi @Dawson-Hsu 
Try building it in the app directory, in your case F:\dev\west-workspace\bms-firmware\app...
This is treated as a freestanding application since it is outside the zephyr workspace, refer to this: https://docs.zephyrproject.org/latest/develop/application/index.html

Dawson-Hsu, 2025-05-30:
1、According to your guidance, a new problem has emerged. How to fix it?

[image]                                                                               

cmake_log show this error 

[image]