LibreSolar/bms-firmware #50: Control flow of code using thingset.

Ananyaaynana, 2024-02-03:
Set an infinite loop that waits for a variable to be set through thingset in the main function. 
This is done to control the flow of the main function using thingset.
Below is the code snippet:
[image]

The program does not enter the infinite loop, and wait for the value to be changed via thingset but proceeds to the next function. The output is:
[image]

Not able to understand why this is happening.

martinjaeger, 2024-02-04:
I don't understand what you are trying to achieve. And I don't see how this is related to ThingSet. How is the `useDefaults` variable changed in the background?

Can you please push a complete sample to a git repo?

Ananyaaynana, 2024-02-05:
The variable "UseDefault" is defined in the bms structure like this:
[image]

In thingset  the "UseDefault" value is defined in a group called "CellChar" like this:
[image]

The flow of the main function should depend on the value of "bms.UseDefault" which is set through thingset. If "bms.UseDefault" is set, the main function should follow one of two paths; if "bms.UseDefault" is not set, the other path is chosen. This should ideally be implemented through IPC but thingset does not support IPC.

Hence, the following code is implemented:
[image]
 where UNDEFINED_STATE is a macro containing the value -1. This code is introduced after the bms_configure(&bms); function is called in the main(void) function.

If "bms.UseDefault" is set to 1 in the terminal using thingset it breaks from the loop. This is a test code to check if this approach of using thingset to determine the flow of the main function is viable. 

The above code was changed to include a ten-second delay to set the "wUseDefaultFlag" in the terminal:
[image]

On setting the "wUseDefaultFlag" to 1 in the terminal after a ":84" code the board crashes:
[image]

Is it possible that the problem is happening due to the shell not being initialised at that point in the main() code?

martinjaeger, 2024-02-05:
What do you mean by "ThingSet does not support IPC"? You can assign callbacks to detect when variables are changed in a particular group and use any IPC mechanism available in Zephyr.

Not sure about the fault. I doubt it's the shell, but I can only guess if you don't provide the complete code.