LibreSolar/bms-firmware #59: I2C error

Ananyaaynana, 2024-02-29:
Hardware used:
Using a custom board with the same I2C configuration as the 0.4v bms_c1 board. GPIO8 is SDA and GPIO9 is SCL.
[image]

Description of the issue:
We are using the bms_c1 firmware at this commit https://github.com/LibreSolar/bms-firmware/tree/cb89c75b74e85a036b7c6243424e31c68b3b3018. An app.overlay file that updates the I2C pins to match the I2C configuration of the 0.4v bms_c1 board.
A few modifications have been done to the BMS firmware such as recovery for overcurrent in charge and discharge and short circuit. We want to test these modifications in the lab. 

When the firmware is flashed on the custom board getting this error:
[image]

Probing test points 12 and 13 revealed no observable signal waveforms on SDA and SCL lines. 

app.overlay:
[image]

Impact:
Not able to test the firmware with recovery modifications.

Zephyr version: v3.4
    commit: b7e659cdd3088609c1e0b9d7f925a811c006f6bd

martinjaeger, 2024-02-29:
Did you also try if it works with most recent firmware commit in main? It contains several improvements. Maybe it solves your issue.

martinjaeger, 2024-02-29:
Ok. The overlay looks correct. Did you make sure it's picked up by the build system? (see output at the beginning of `west build` command)

Ananyaaynana, 2024-02-29:
[image]
Yes, it has picked the app.overlay file during the build.

Ananyaaynana, 2024-03-01:
We tried to debug the issue but it was unsuccessful. So we are integrating the recovery modifications in the latest commit but we hit a few roadblocks. I will open another thread to explain the issue.

closed 2024-03-01 by Ananyaaynana