LibreSolar/bms-firmware #61: Questions Regarding using Multifunctional Pin controls as thermistors

Ananyaaynana, 2024-03-12:
I'm reaching out with a few questions regarding the Battery Management System (BMS) configuration in the codebase.

Shunt Temperature Measurement:
        I'm exploring the bms_c1.dts file to understand shunt temperature measurement. I couldn't locate the configuration for the DFETOFF pin. Is shunt temperature currently measured and used to trigger alerts in the BMS system?

Default Values in BQ769X2_INIT Macro:
        I'm curious about the configuration of temperature and discharge pins in the BQ769X2_INIT macro. While ts1_pin_config has a default value of 0x07, ts3_pin_config and dchg_pin_config seemingly don't. Could you explain the thinking behind this distinction?

DCHG Pin Configuration:
In the code, dchg_pin_config is set to the default value (0x07). The BQ769X2 datasheet (section 6.4) recommends 0x0F for FET temperature measurement. I'm curious to understand the reasoning behind using the default value in this case.

Understanding these points would be very helpful.

martinjaeger, 2024-03-12:
> Shunt Temperature Measurement: I'm exploring the bms_c1.dts file to understand shunt temperature measurement. I couldn't locate the configuration for the DFETOFF pin. Is shunt temperature currently measured and used to trigger alerts in the BMS system?

The shunt temperature measurement was only added for BMS C1 v0.4 board. It is not yet supported by the firmware.

> Default Values in BQ769X2_INIT Macro: I'm curious about the configuration of temperature and discharge pins in the BQ769X2_INIT macro. While ts1_pin_config has a default value of 0x07, ts3_pin_config and dchg_pin_config seemingly don't. Could you explain the thinking behind this distinction?

Yes, that's to mimic the default behavior of the chip. The actual desired values of used sensors should be set via board .dts file.

> DCHG Pin Configuration: In the code, dchg_pin_config is set to the default value (0x07). The BQ769X2 datasheet (section 6.4) recommends 0x0F for FET temperature measurement. I'm curious to understand the reasoning behind using the default value in this case.

Yep, that's wrong. The configuration for the FET temperature sensor should indeed be `0xF`. Thanks for noticing. If you like, feel free to submit a PR with a fix.

Ananyaaynana, 2024-03-13:
Thank you for the reply.
Sure, I will submit a PR request with the fix.

PR #64 (2024-03-13): Changed dchg-pin-config value 
Changed dchg-pin-config value from 0x07 to 0x0F to configure this pin for measuring the temperature of FETs on the board. See section 6.4 of bq76952 technical ref. Closes #61