LibreSolar/bms-firmware #66: Warning when sending CAN data

Ananyaaynana, 2024-03-16:
Testing the CAN interface on a custom board similar to 0.4v of BMS C1.
On the serial interface getting this as the output:
[image]
With a warning saying that unable to send CAN frame since it is too large. 
The output of the CAN software was this:
[image]
What can be the cause of this warning?

martinjaeger, 2024-03-16:
That's because the voltages and temperatures don't fit into a single classical CAN frame. With the new ThingSet SDK packetized reports we send all data items in a multi-frame report to make this work. This was not yet included in the BMS firmware.

I have just merged #65 to include it. Please double-check if the warning has gone.

Ananyaaynana, 2024-03-18:
Using commit: https://github.com/LibreSolar/bms-firmware/tree/187d315c7acb94628d0c467649940d386e8e79c5 at this commit the board is not able to send data through CAN. Output at the serial terminal:
[image]
Data was not received in the CAN software dashboard.

Using commit: https://github.com/LibreSolar/bms-firmware/tree/50b5bc5bdae9ee7609a8f9edb43ee9fb062481d9 at this commit the board can send data through CAN. Output at the serial terminal:
[image]
Data received in the CAN software dashboard.

martinjaeger, 2024-03-19:
Sorry, I forgot to enable one additional required Kconfig. Fixed in `main` now. See this commit: https://github.com/LibreSolar/bms-firmware/commit/fb078cc4b56179d12973b2ab40afc57f59054fa0

Ananyaaynana, 2024-03-19:
Data received on the CAN software dashboard.
Output on serial terminal:
[image]
Thank you for the help!

closed 2024-03-19 by Ananyaaynana