LibreSolar/bms-firmware #68: Error flag 1 on connecting the board to a battery pack

Ananyaaynana, 2024-04-04:
Description:
When the BMS is brought up an error flag 1 is always present. Error flag 1 indicated cell Undervoltage.  None of the cells are below the value of 2.5V for LFP cell. Here is a picture on the thingset app:
[image]

Once I run the `west espressif monitor` the error flag is cleared like in this picture:
[image]

Double-checked the connections and it looked fine. This happens every time the BMS is brought up.

martinjaeger, 2024-04-06:
I can see that the last cell measurement shows `0.000` at the bottom of your screenshot. How many cells do you have connected? And how does your `used-cell-channels` parameter in Devicetree look like?

Ananyaaynana, 2024-04-08:
15 cells are connected in series.  `used-cell-channels` is 0xFFFF.

martinjaeger, 2024-04-20:
I have checked here (with 4 cells connected) and I cannot reproduce the error. In my case all cells from 5 to 16 show 0.0V, but the number of cells seems to be detected correctly.