LibreSolar/bms-firmware #69: How to enable Wi-Fi on BMS C1

ananya-wattsnext, 2024-05-15:
I want to enable wifi to transmit data to the dashboard. I am using commit: 3bd1fbab1f01fb142f51d2e3b791c772b77745f0. 
What are the Kconfig options to enable wifi?

martinjaeger, 2024-05-15:
I haven't tried with the BMS, yet, but you can look at [this configuration](https://github.com/ThingSet/thingset-zephyr-sdk/blob/main/samples/counter/wifi_websocket.conf) to enable Wifi and ThingSet over WebSocket.

The amount of RAM may not be sufficient for Bluetooth, WiFi and WebSockets with TLS at the same time, so you may have to disable Bluetooth or fine-tune stack sizes.

ananya-wattsnext, 2024-05-15:
[image]

The sample is not building for bms_c1@0.4 but I am able to build and execute this sample for the esp32c3 devkitm board.