LibreSolar/bms-firmware #76: why alert_mask variable clear at function entry?

firepooh, 2024-07-16:
hi~

may i question your bms code. 

why alert_mask variable clear at function entry?

after exe "ic_conf->alert_mask = 0" no more effect code 

any here something else hidden function?

https://github.com/LibreSolar/bms-firmware/blob/ce8b8766e49f4c986b83eab838e92be2ccf91f74/drivers/bms_ic/bq769x2/bq769x2.c#L414

martinjaeger, 2024-07-16:
If the alert mask is not cleared, previously set flags will be maintained, which is not desired.

Vadiml1024, 2024-11-15:
But this means that the alerts will be ALWAYS disabled as
expressions  xxx = !!(ic->data.alert_mask & ZZZ)   will be ALWAYS false

martinjaeger, 2024-11-16:
Sorry for not taking a better look. You are right @firepooh, this is a bug. It's now fixed in #81.

PR #81 (2024-11-16): bq769x2: Fix alert_mask configuration write-back
Fixes #80
Fixes #76
closed 2024-11-16 by martinjaeger